mukaschultze/ha-must-inverter #8: error reading modbus data at address ******

omarmeleis, 2024-02-02:
### System Health details

## System Information

version | core-2024.1.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.71-haos
arch | x86_64
timezone | Africa/Cairo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.34.0
Stage | running
Available Repositories | 1382
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | July 14, 2024 at 3:00 AM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | us-east-1-6.ui.nabu.casa
certificate_status | ready
instance_id | 087c330f972d4b5894898f6f7cff68ee
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.4
-- | --
update_channel | stable
supervisor_version | supervisor-2023.12.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 468.7 GB
disk_used | 40.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.0), Log Viewer (0.17.0), Check Home Assistant configuration (3.11.0), File editor (5.7.0), Z-Wave JS (0.4.3), AirCast (4.2.0), Home Assistant Google Drive Backup (0.112.1), ESPHome (2023.12.9), Advanced SSH & Web Terminal (17.0.5), AdGuard Home (5.0.2), Samba share (12.2.0), DeepStack (2021.09.1), Exadel CompreFace (1.1.0), Filebrowser (2.23.0_12), Portainer (2.19.4-3), Double Take (1.13.11.8), Network UPS Tools (0.13.0), InfluxDB (5.0.0), Frigate (0.13.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 5
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 24, 2024 at 6:47 AM
-- | --
current_recorder_run | February 1, 2024 at 11:25 PM
estimated_db_size | 321.76 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.4 (a4a8c5f)
-- | --
cloud_online | 3 / 4
local_online | 3 / 3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I am always getting the following error after a few minutes of successfully connecting and getting data of the inverter

Logger: custom_components.must_inverter
Source: custom_components/must_inverter/__init__.py:211
Integration: Must Inverter ([documentation](https://github.com/mukaschultze/ha-must-inverter), [issues](https://github.com/mukaschultze/ha-must-inverter/issues))
First occurred: February 1, 2024 at 11:26:30 PM (8555 occurrences)
Last logged: 1:23:45 PM

error reading modbus data at address 20001
error reading modbus data at address 20101
error reading modbus data at address 25201
error reading modbus data at address 15201
error reading modbus data at address 10101
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 211, in read_modbus_data
    response = await self._client.read_holding_registers(address=register[0], count=register[1], slave=0x04)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 147, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB0:0]

I have used different methods and different adaptors, but getting the same issue
When i reload the integration, it fixes it for a few minutes, then stops again so I have to reload the integration again

I have made an automation to automatically reload the integration when getting this error from the log

I have a must PV1800 3k inverter, now connected through the RJ 45 port to a USB to RS485 Converter Adapter then to my HA instance

### Reproduction steps

1. Successfully connected to the inverter via RS485 and serial
2.After a few minutes, i get the error reading modbus data at address 20001 etc
3.have to reload the integration for it to pull data again from the inverter 
...

### Debug logs

```Text
Logger: custom_components.must_inverter
Source: custom_components/must_inverter/__init__.py:211
Integration: Must Inverter (documentation, issues)
First occurred: February 1, 2024 at 11:26:30 PM (8555 occurrences)
Last logged: 1:23:45 PM

error reading modbus data at address 20001
error reading modbus data at address 20101
error reading modbus data at address 25201
error reading modbus data at address 15201
error reading modbus data at address 10101
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 211, in read_modbus_data
    response = await self._client.read_holding_registers(address=register[0], count=register[1], slave=0x04)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 147, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusSerialClient /dev/ttyUSB0:0]
```

mukaschultze, 2024-02-02:
@omarmeleis did you change any of the default connection settings? I also have the PV1800 and noticed that the serial interface dies when too many commands are being sent in a short period of time. The default settings work fine for my inverter.

Try increasing the `Reconnect Delay` and `Max Reconnect Delay` a bit and see if it stops this error. You might need to remove and add the integration again to change these, I'm still developing it and the configuration form is not fully ready yet.

omarmeleis, 2024-02-02:
I must say that this is  one of the best ways i have found to pull data from the must inverter,,,,, so Cheers for your amazing work :D

No i have not changed anything in the default connection settings,,, moreover, I have disabled most of the entities and enabled only around 10

Any recommendations regarding the values of the connection settings,,, increase by how much ??

mukaschultze, 2024-02-02:
Try setting the Delay to 1.0 and the Max Delay to 1.5, also increase the Timeout to something like 5 so the increased delay won't trigger it.

mukaschultze, 2024-02-04:
@omarmeleis I released a new version that tries to reconnect between reads. Let me know if this works better for you.

This version will possibly break your existing configuration and you'll need to reconfigure. I'm still learning how HA custom integrations work :grimacing:

omarmeleis, 2024-02-04:
Thank u for the amazing work and effort....
I will try the new version and give u an update

omarmeleis, 2024-02-04:
I have tried the new version, but it still has the same error, but some times gives me a different one when reloading the integration 

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:507
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 7:53:50 PM (1 occurrences)
Last logged: 7:53:50 PM

Error adding entities for domain sensor with platform must_inverter
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 614, in _async_add_entity
    if self.config_entry and (device_info := entity.device_info):
                                             ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/sensor.py", line 66, in device_info
    return self._inverter._device_info()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/__init__.py", line 253, in _device_info
    "identifiers": {(DOMAIN, self.data["InverterMachineType"])},
                             ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'InverterMachineType'

ogger: homeassistant.components.binary_sensor
Source: helpers/entity_platform.py:507
Integration: Binary sensor ([documentation](https://www.home-assistant.io/integrations/binary_sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+binary_sensor%22))
First occurred: 7:53:50 PM (1 occurrences)
Last logged: 7:53:50 PM

Error adding entities for domain binary_sensor with platform must_inverter
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 614, in _async_add_entity
    if self.config_entry and (device_info := entity.device_info):
                                             ^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/binary_sensor.py", line 55, in device_info
    return self._inverter._device_info()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/__init__.py", line 253, in _device_info
    "identifiers": {(DOMAIN, self.data["InverterMachineType"])},
                             ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'InverterMachineType'

omarmeleis, 2024-02-04:
Also getting this error

This error originated from a custom integration.

Logger: py.warnings
Source: custom_components/must_inverter/__init__.py:228
Integration: Must Inverter (documentation, issues)
First occurred: 7:46:48 PM (3 occurrences)
Last logged: 8:23:31 PM

/config/custom_components/must_inverter/__init__.py:228: RuntimeWarning: coroutine 'MustInverter._check_and_reopen' was never awaited if not self._check_and_reopen():

PraveenAlexis, 2024-02-05:
I have the same issue, and thanks for the amazing work you have done :)

muhrin0102, 2024-03-19:
Good day
I would like to have a basic question, how do you solve the hardware connection between the Home assistant and the Must inverter?

mukaschultze, 2024-03-19:
@muhrin0102 Mine comes with a built-in USB port (that's just a USB to serial converter internally), I've used this port for some time. After a while, I bought a serial to wifi module on AliExpress so I could move my server away from the inverter and it worked nicely.

[This is the one I bought, it's din rail mounted and there's both an AC and DC version](https://pt.aliexpress.com/item/4001114113365.html) but there's also [a cheaper alternative](https://pt.aliexpress.com/item/32916128353.html), from what I've searched both use the same board

muhrin0102, 2024-03-22:
> [mukaschultze](/mukaschultze)
Hi
Thank you very much for the advice and recommendations. I currently have the inverter connected via USB cable, I'm going to order the RS485 to Wifi converters recommended by you. I have one more problem, I use three inverters, one for each phase. I connected the first one in order (sometimes the USB connection goes crazy), when I connect the second one, the problem arises that it generates the same entities for both inverters. It would be necessary to insert here ("name": self.data["InverterMachineType"]) some unique identifier for each inverter. Once again, thank you very much for the answer

mukaschultze, 2024-03-23:
@muhrin0102 can you create a new issue and paste your debug logs there? I suspect there is information about the multi-phase setup in the Modbus data but mine doesn't have this feature so I'm not sure how it's presented.

Spais07, 2024-05-05:
Hello.
I have the same issue.
After few minutes of working it stop reading any parameters.
I ill connect my inverter to HomeAssistant through USB-A - USB-B interface.
How can be fixed this?
Must ph1800

klbsss, 2024-05-30:
same issue

NothingWeAre, 2024-06-16:
It looks like some kind of leak.

In my case HA stops communicating with inverter (PH1800) after 2 ~ 3 hours with errors
`Modbus Error: [Input/Output] ERROR: No response received after 3 retries` and  `error reading modbus data at address`

As a temporary solution I have setup scheduled restart of MUST integration, every 2 hours. 
And over 3 days there was no errors in the log

PS.
https://github.com/wills106/homeassistant-solax-modbus/issues/801
This is another integration that relies on `pymodbus`.  and looks like having related problem.

NothingWeAre, 2024-06-16:
May be related
https://github.com/pymodbus-dev/pymodbus/issues/2193

mr-p666, 2024-07-10:
After updating to HA 2024.7.2 and thus to pymodbus 3.6.9 my Modbus integration is finally working again.
I already tried a restart to make sure it still works after that and at the moment everything works as before the update to 2024.3.3.
Thanks a lot @janiversen!

mukaschultze, 2025-02-19:
Not sure it's related to #64, I had the same issue without having multiple devices reading from the same modbus slave.

I think it was a bugged version of pymodbus since people stopped reporting connection issues shortly after I updated the version used by this integration.

Anyway, I'll close this issue anyway since there hasn't been any activity recently. If anyone else is having the same problem please open a new issue.