mukaschultze/ha-must-inverter #20: Support to wifi device https://pt.aliexpress.com/item/1005003231633440.html?spm=a2g0o.order_list.order_list_main.5.4426caa4cG0Xmp&gatewayAdapt=glo2bra

alessandrokornowski, 2024-03-26:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

This is suposed to be MUST oficial wifi device.
SMART ESS is really poor as management and monitoring app.
Home Assistant should be better to monitor device

### Describe the solution you'd like

Set  host and port to use this adapter to monitor MUST PV-1800 over wifi
Home Assistant should be better to monitor device

### Describe alternatives you've considered

SMART ESS APP is working fine

### Additional context

If remote access is needed to this device over internet, i can arrange a NAT to ip address of developer to conect it.

mukaschultze, 2024-03-26:
@alessandrokornowski hello fellow Brazilian! The AliExpress description says it's a Modbus-TCP, so it should work with the current version of the integration, you just need to figure out its IP address and port.

[image]

youmax01, 2024-05-22:
HI everyone

The wifi dongle of the inverters must only communicate with the cloud service www.shinemonitor.com

it is not possible to intercept modbus messages locally. The option for those of us who have this dongle and the inverter far from the home assistant equipment is to read the information that the WiFi dongle sends every 5 minutes to the cloud service

Here is an option that works well, changing some parameters, it would be very good to add the "online" option
I am not very good at programming but I have found how to make it work so that it extracts the data from the web page

To do this, the following data is needed, which is taken from the website:

# Shinemonitor settings
base_url = 'http://android.shinemonitor.com/public/'
usr = ''  # Username
pwd = ''  # Password
company_key = ''  # Company key. Obtained from portal
plant_id = ''  # Plant id (Power station ID). Obtained from portal.
pn = ''  # Datalogger PN number. Obtained from portal
sn = ''  # Device serial number. Obtained from portal
devcode = ''  # Device coding. Obtained from portal

https://github.com/itsmariodias/shinemonitor-reporter-mqtt

youmax01, 2024-05-23:
http:// http://web.shinemonitor.com/public web page send JSON like this:

{"err":0,"desc":"ERR_NONE","dat":[{"title":"id","val":"7710e23aa1c51deb43c4df617e6eb472"},{"title":"Timestamp","val":"2024-05-23 08:38:51"},{"title":"Battery Voltage","unit":"V","val":"48.3"},{"title":"PV Voltage","unit":"V","val":"52.2"},{"title":"Inverter Voltage","unit":"V","val":"225.0"},{"title":"Batt Current","unit":"A","val":"0"},{"title":"Charger Current","unit":"A","val":"0.1"},{"title":"Charger Power","unit":"W","val":"10"},{"title":"PLoad","unit":"W","val":"192"},{"title":"PGrid","unit":"W","val":"-203"},{"title":"work state","val":"Grid-Tie"},{"title":"rated power","unit":"W","val":"5500"},{"title":"Grid Voltage","unit":"V","val":"227.2"},{"title":"PInverter","unit":"W","val":"0"},{"title":"Accumulated Sell Power","unit":"kWh","val":"0"},{"title":"Accumulated Load Power","unit":"kWh","val":"11087"},{"title":"Accumulated Self_Use Power","unit":"kWh","val":"3001"},{"title":"charger work enable","val":"ON"},{"title":"Accumulated PV Power","unit":"kWh","val":"4194.70"}]}

alessandrokornowski, 2024-05-23:
Hi @youmax01 
I´ve been trying for some time port scan Dongle IP address and find no udp or tcp port opened, so that´s why @mukaschultze didnt worked.
I´m wondering how SmartESS installed on cel phone is able to discover wifi device on LAN. For this some port should be open to discover it. nmap and fing didn´t found any open port on dongle.

By the way... SmartESS user and password that is in use on my cel phone and correctly monitoring inverter is not known by shinemonitor.com portal.
Is needed to change some dongle option to use shinemonitor instead of SmartESS backend?

youmax01, 2024-05-23:
> Hi @youmax01 I´ve been trying for some time port scan Dongle IP address and find no udp or tcp port opened, so that´s why @mukaschultze didnt worked. I´m wondering how SmartESS installed on cel [...]

Hi Alessandro

The wifi dongle only connects to the shinemonitor server, and the application takes the data from there, not from the dongle.
You can test by disabling internet connectivity, it should not work at all

The first configuration that the app makes towards the dongle is only to register it in the Wi-Fi network and add the serial number of the dongle to the shinemonitor server.

The only option is to take the data from there in online mode.

From what I was able to investigate, the server does a handshake with the dongle, without that the device does not start transmitting the data

youmax01, 2024-05-23:
another good option is with ESPHOME , ESP32 board + rs485 board, read the data and sent to mqtt to home assistant wireless

[image]

alex-dokienko, 2024-07-13:
@youmax01 wonder if you have some article or video which describe how this can be done in details for some newbies

LionBS76, 2024-09-17:
> another good option is with ESPHOME , ESP32 board + rs485 board, read the data and sent to mqtt to home assistant wireless
> 
> ![1_J3x1SaSq1cEe5UincnCdUA](https://private-user-images.githubusercontent.com/35151341/333333043-ac16025f-a573-41e2-9188-7904c4e46312.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY1ODM0MzAsIm5iZiI6MTcyNjU4MzEzMCwicGF0aCI6Ii8zNTE1MTM0MS8zMzMzMzMwNDMtYWMxNjAyNWYtYTU3My00MWUyLTkxODgtNzkwNGM0ZTQ2MzEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE3VDE0MjUzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViZGM5MjFmZDQ0ZDQ2OGQ3ODE0N2I2NWEwYmZhNGRjM2FkYjdkMmM5NjE4OGU4MjU3ZTM0YzQxZDI3NzZjMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6MxqeZGUN3IWVzxo7-DhTXwPlAEPmghEKemHLNqyfmk)

Why use RS485 ? Must inverter have USB Type-B (RS232) communication port. 
I think it is possible to use esp8266 + uart-to-usb ?

mukaschultze, 2024-09-17:
@LionBS76 both the USB and the RS485 ports use modbus as the communication protocol, so the layout you're thinking should work just fine.

bernstein7, 2024-09-22:
> Why use RS485 ? Must inverter have USB Type-B (RS232) communication port. I think it is possible to use esp8266 + uart-to-usb ?

but you'd still have to convert RS232 to ttl right? so that'd be pretty much the same layout as with RS485
[image]

bernstein7, 2024-09-22:
> @LionBS76 both the USB and the RS485 ports use modbus as the communication protocol, so the layout you're thinking should work just fine.

Is the MQTT broker the only way to feed data from ESP into HA or this integration also can receive data from ESP and feed it to HA? Thanks!

mukaschultze, 2024-09-22:
> Is the MQTT broker the only way to feed data from ESP into HA or this integration also can receive data from ESP and feed it to HA? Thanks!

This integration has nothing to do with MQTT or ESP32. That said, it'll work with anything that provides a Modbus connection via TCP, UDP, or USB. See my comment on this other issue: https://github.com/mukaschultze/ha-must-inverter/issues/36#issuecomment-2219485816