mukaschultze/ha-must-inverter #23: Speedup modbus response

daimonkor, 2024-04-22:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hello, pleae answer. How I can change scna_interval to less?. I see like 10-13 seconds between responses. Thanks a lot for help.

### Describe the solution you'd like

 Missed

### Describe alternatives you've considered

 Missed

### Additional context

 Missed

mukaschultze, 2024-04-29:
@daimonkor I've added a new configuration that allows you to change the scan interval

Closed by 11d4202cf6924718f2efa883ecb38b197e78d76e

closed 2024-04-29 by mukaschultze

daimonkor, 2024-04-29:
Thanks a lot, it works.