mukaschultze/ha-must-inverter #24: ModBus USB serial "Retries" parameter is depreciated - should remove

OwlBawl, 2024-04-25:
### System Health details

ModBus USB serial "Retries" parameter is depreciated - should remove

And have some problem with "Reset the accumulated data"
it doesn't work somewhy...  

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

ModBus USB serial "Retries" parameter is depreciated - should remove

And have some problem with "Reset the accumulated data"
it doesn't work somewhy...  

### Reproduction steps

1.
2.
3.
...

### Debug logs

```Text
1.
2.
3.
...
```

mukaschultze, 2024-04-29:
@OwlBawl  The 20213 register only removes the accumulated data for the inverter, not for the charger. I have added it as a new entity

As for the retries being deprecated, where did you see that? I couldn't find anything in the Pymodbus' documentation and source code.

Closed by 4ecae2888aa6c2cc43bceeefdfb2dc64e30b7c40

closed 2024-04-29 by mukaschultze

OwlBawl, 2024-04-29:
Thank you for 20213 register update.

As of Retries - maybe my fault. Used native HA modbus for my Must inverter before found your more complete implementation (thank you) and after earlier updates - had to remove this parameter from config.yaml because of update notification (changelog). They deleted it in description also https://www.home-assistant.io/integrations/modbus/