mukaschultze/ha-must-inverter #25: Add new property

daimonkor, 2024-04-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

asdasdaa

### Describe the solution you'd like

asdaaaa

### Describe alternatives you've considered

adaaaa

### Additional context

aasdsd

mukaschultze, 2024-04-30:
@daimonkor is there a particular reason why you need this data? I don't know how I could present this info in a way that it would be useful.

Is that anything in that register that is not available in other register (for example the work status and relays status)?

daimonkor, 2024-04-30:
@mukaschultze  as I understood the information contains data about flow enegry - when batery are charging by grid or solar. In my opinion currentr infirmation is usefull to calculate daily usage solar to charge battery and correct calculation usage solar to load. thanks a lot for help.