mukaschultze/ha-must-inverter #36: How to connect inverter physically to Home Assistant OS (run on Raspberry Pi) to use your integration

misterscv, 2024-07-10:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Sorry for a dumb question im noob :). I put it here, cos i do not know how to ask a question in other way.
 
How to connect inverter physically to Home Assistant OS (run on Raspberry Pi) to use your integration. Should i use official usb wifi plug, or use some cables such as rj45 or usb cable to connect inverter to HA. Can you explane this, please. Thanks in advance

### Describe the solution you'd like

Sorry for a dumb question im noob :). I put it here, cos i do not know how to ask a question in other way.

### Describe alternatives you've considered

Sorry for a dumb question im noob :). I put it here, cos i do not know how to ask a question in other way.

### Additional context

Sorry for a dumb question im noob :). I put it here, cos i do not know how to ask a question in other way.

mukaschultze, 2024-07-10:
What's your inverter model? 

There are two ways of connecting the inverter:

### USB (easy to set up)

My inverter (PV18-3024 VPM) comes with a USB cable, I'm assuming yours does too. You can use this cable to connect the inverter to a PC using port 8 of the screenshot below (or Raspberry in your case).

After you connect the USB cable the inverter will show up as a serial device usually on `/dev/ttyUSB0` path. All the correct settings will be auto-completed when you try to add the device to Home Assistant.

This is by far the easiest way to set up, it should work out of the box.

[image]

### Using a serial to Wi-Fi converter (advanced set up)

Alternatively, you can buy a generic serial to Wi-Fi converter and use the RJ45 (port 9 in the screenshot) to connect your device to your home network.

But be aware that this is not as simple as connecting an ethernet cable to the inverter, it requires you to crimp your own RJ45 cable using the pinout from your inverter's manual. It also requires a fair bit of configuration, you'll need to have a bit of understanding of how serial devices work.

**The official wifi plug is NOT supported by the integration since it uses some proprietary cloud service.**

This is the serial to wifi converter I've been using for the last 5 months on my setup: https://pt.aliexpress.com/item/4001114113365.html

Let me know if this clears up how to setup the integration a bit so I can add better instructions to the readme file

misterscv, 2024-07-10:
My inverter is HBP1800 (HBP18-5248 HM). It has usb port and cabel included.

I connected USB to my Raspberry. It shows as
[image]
I cannot understand what to type in the first field in my case
[image]

misterscv, 2024-07-10:
It works! :)

i typed:

/dev/ttyUSB0

and it works!

Thanks

misterscv, 2024-07-10:
My battery is 48v. So all voltage controls are wrong. But Sensors Monitor works! That is what i needed.

It would be nice if all data were correct :) But i m glad that i can use sensors.

Thanks for integration

mukaschultze, 2024-07-10:
I've added the default placeholder to that field, hopefully, it solves the confusion for other people 7fa4be3c45370cb52af736a7f6d9f6442a2e3a5c

misterscv, 2024-07-11:
Hi.  I have another question.  does the inverter know about the capacity and level of the connected battery in wh or %.   It would be great to have such information as sensors

alex-dokienko, 2024-07-13:
hello, wonder if you can help, I've 5kWh inverter + 48V Must battery, I managed to connect via USB using your integration, big thank you for effort on building and sharing this integration.
Currently I would like to get  battery capacity level, but I couldn't find any metrics related to that, only current batt power.

I know battery can be connected via CAN cable to inverter (maybe in that case it can give more options ?) but it's not an option for me as in that case it tries to use 80A charging which I can't use due to existing setup (not so good wires)

Is there any way how I can get that information ? Really appreciate any help or guideness.

misterscv, 2024-07-13:
Hi, Alex. I asked [mukaschultze](https://github.com/mukaschultze) for the same question :). I have the HBP18-5248 HM station. It has inverter and 48v, 5kWh battery. I believe, you need to connect battery and inverter via cable to CAN and COM ports. It would not give capacity information, but inverter will work properly with battery. By default battery charges with 30A, and you can change it to 60A max. I think you mix up max charge settings from battery and solar panels togather, which is 80A max.

As for capacity sensor and battery level sensor, i created my own template sensor in HA. From existing data from integration. it works but not as good as it should :)

It would be great if it possible some how to add this information into the integration itself. 
And big plus if integration would understand voltage of battery and give propper controls for the battery. Integration knows about battery, but has controls from 24v battery. [mukaschultze](https://github.com/mukaschultze) what do you think.

alex-dokienko, 2024-07-13:
hi @misterscv, thanks for feedback.
As for connecting battery and inverter with CAN cable I'm pretty sure it resets/ignore your charging limits, as I did all setup to use charging 30A, but once I used cable, setting was resetted by default to 80A (as that what my battery can take), I tried to change charging back to 30A, but I think it can't store it or just reset every time if I do switch off inverter, didn't want really to play around too much with that option as my wires really can't take such current + I also found some other guy in Interent who also experienced same issue as me, also I don't have any solar panels, using my inverter in appartment. 

In my case I'm using using as output switch with energy meter function, but yes it's also working not so great, in my situation I charge battery once at night (due to lower cost), so it's become hectic to keep track current capacity especially if it's charged during day (in some cases when we get long loadshedding schedule).

mukaschultze, 2024-07-13:
> Hi. I have another question. does the inverter know about the capacity and level of the connected battery in wh or %. It would be great to have such information as sensors

@misterscv @alex-dokienko  the inverter doesn't provide any sort of battery percentage or remaining capacity. You would need some sort of coulomb counter to keep track of the SOC. 

The inverter LCD display uses a heuristic to determine the capacity, I guess it's possible to implement this same heuristic using HA templates, but keep in mind that it's extremely inaccurate.

[image]

> As for capacity sensor and battery level sensor, i created my own template sensor in HA. From existing data from integration. it works but not as good as it should :)

@misterscv can you paste your template here? Might be helpful to some folks.

> Integration knows about battery, but has controls from 24v battery. [mukaschultze](https://github.com/mukaschultze) what do you think.

I can try to do that, only reason I haven't done it yet is because nobody complained :upside_down_face:

misterscv, 2024-07-14:
I have some problems with integration and do not know how to fix it. All integration sensors stuck at some moment after long peropd of work and do not refresh their data. For example, sensors can work for 5-8 hours and then stuck. Reload integration does not help. After reload integration all sensors become unavailable. Only restart home assistant helps to make the integration to work.
[mukaschultze](https://github.com/mukaschultze) do you have any thoughts. Have you faced this problem?

mukaschultze, 2024-07-14:
> [mukaschultze](https://github.com/mukaschultze) do you have any thoughts. Have you faced this problem?

Yes, increasing the scan interval helps. 15 seconds works for me, anything less than that freezes the inverter after a few hours.

alex-dokienko, 2024-07-14:
actually I got similar issue today, well my inverter is stopping/starting based on automation, so after inverter start again, automation stop receiving values, I'd imagine after some retries it timeouts and not reconnects after that.

LionBS76, 2024-09-16:
> What's your inverter model?
> 
> There are two ways of connecting the inverter:
> 
> ### USB (easy to set up)
[skip]
> ### Using a serial to Wi-Fi converter (advanced set up)
> Alternatively, you can buy a generic serial to Wi-Fi converter and use the RJ45 (port 9 in the screenshot) to connect your device to your home [...]

Hello!
About Serial-to-WiFi converter to connect to inverter remotly.
Regular USB interface on inverter using RS232 protocol. 
So why not to using the same RS232 protool not RS485 (RJ-45) to connect to inverter remotly?
Is there any RS232-to-WiFi converters ?

Thanks!

LionBS76, 2024-09-19:
>
> There are two ways of connecting the inverter:
> 
> ### USB (easy to set up)
> My inverter (PV18-3024 VPM) comes with a USB cable, I'm assuming yours does too. You can use this cable to connect the inverter to a PC using port 8 of the screenshot below (or Raspberry in your case).

Hello !
Is it possible to connect to USB WiFi port (port 12 on screenshot) ?
If I try to connect PC to this 12 port Windows doesn't recognise this port properly. 
USB device id in this port 12 is USB Hub , not as USB Device like in port 8.

So the main question, is it possible to use port 12 (USB WiFi Port) to connect to Home Assistant with your integration ?

mukaschultze, 2024-09-19:
> So the main question, is it possible to use port 12 (USB WiFi Port) to connect to Home Assistant with your integration ?

@LionBS76 as far as I know you can't. That USB-A port most likely is treated as a USB host and not a USB device (as the USB-B is) so your computer won't recognize it.

Alex165-cmd, 2024-09-23:
Greetings. Which of the sensors is responsible for the generated energy of the solar cells? To add it to the energy panel. Thank you.

mukaschultze, 2024-09-23:
> Greetings. Which of the sensors is responsible for the generated energy of the solar cells? To add it to the energy panel. Thank you.

@Alex165-cmd `accumulated_power` should work with the energy dashboard out of the box. However, in my testing the inverter was not super accurate when measuring the total power, if you want better precision I'd recommend creating an integration sensor using the `charger_power` as the source

Alex165-cmd, 2024-09-25:
> > вітаю. Який із датчиків відповідає за вироблену енергію сонячних елементів? Щоб додати його до енергетичної панелі. дякую
> 
> @Alex165-cmd `accumulated_power`повинен працювати з енергетичною приладовою панеллю з коробки. Однак під час мого тестування інвертор не був дуже точним під час вимірювання загальної потужності. Якщо вам потрібна краща точність, я б порекомендував створити датчик інтеграції, використовуючи `charger_power`як джерело

Congratulations. Thanks for the answer. Tell me which sensors are responsible for battery charge and battery discharge?

olehsava, 2024-12-19:
Hi everyone,
First of all, I want to thank @mukaschultze for the integration. You did a great job. 
I have PV18-5048 PRO invertor and use RS485->WiFi Protos PW-11 converter. 
The connection between Must inverter and PW-11 converter is quite simple. There is a chart with pins definition in the manual.
[image]
The most difficult part was the Serial Port Configuration. I spent a few hours to find out the correct settings, but the correct Baud Rate (19200) did the trick.
[image]
Also, had to update [const.py](https://github.com/mukaschultze/ha-must-inverter/blob/master/custom_components/must_inverter/const.py) file to match 48v.

sanfilippopablo, 2025-02-18:
Hi! Any pointers on how to set up this integration over USB with a Raspberry Pi far away from the home assistant server? I can connect the inverter via USB to a raspberry pi, but that raspberry pi is not running the home assistant server, but it's on the same network as the home assistant server

mukaschultze, 2025-02-18:
> Hi! Any pointers on how to set up this integration over USB with a Raspberry Pi far away from the home assistant server? I can connect the inverter via USB to a raspberry pi, but that raspberry pi is not running the home assistant server, but it's on the same network as the home assistant server

@sanfilippopablo Check out this repo https://github.com/3cky/mbusd, you can run this on your Raspberry to expose a TCP connection for your HA server to use. I've used it in the past when I was first developing this integration and it works perfectly.

sanfilippopablo, 2025-02-18:
@mukaschultze Thanks! Will try that

alistairts, 2025-04-30:
> official wifi plug is NOT supported

I wish you stated this in your home page. The only reason I purchased the MUST inverter over another brand was because it came with the official wifi plug...

vsylantiev, 2026-01-17:
in dmesg see these lines when reconnect USB Cable (tried all ports):
[  507.012024] usb 1-1.1: new full-speed USB device number 3 using xhci_hcd
[  507.106482] usb 1-1.1: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[  507.106513] usb 1-1.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  507.106528] usb 1-1.1: Product: USB Serial
[  507.150535] usbcore: registered new interface driver ch341
[  507.150644] usbserial: USB Serial support registered for ch341-uart
[  507.150736] ch341 1-1.1:1.0: ch341-uart converter detected
[  507.160105] usb 1-1.1: ch341-uart converter now attached to ttyUSB0
[  555.948258] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[  555.988361] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 2290.272891] usb 1-1.1: USB disconnect, device number 3
[ 2290.274091] usb 1-1.1: failed to send control message: -19
[ 2290.274552] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 2290.274661] ch341 1-1.1:1.0: device disconnected
[ 2294.879819] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[ 2294.973895] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 2294.973920] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2294.973934] usb 1-1.3: Product: USB Serial
[ 2294.978848] ch341 1-1.3:1.0: ch341-uart converter detected
[ 2294.990577] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
[ 2295.412847] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 2295.563138] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 2354.211509] usb 1-1.3: USB disconnect, device number 4
[ 2354.211960] usb 1-1.3: failed to send control message: -19
[ 2354.212430] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 2354.212546] ch341 1-1.3:1.0: device disconnected
[ 2356.305447] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[ 2356.399533] usb 1-1.4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 2356.399560] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2356.399575] usb 1-1.4: Product: USB Serial
[ 2356.404463] ch341 1-1.4:1.0: ch341-uart converter detected
[ 2356.409858] usb 1-1.4: ch341-uart converter now attached to ttyUSB0
[ 2357.270365] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 2357.270852] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 3302.053662] usb 1-1.4: USB disconnect, device number 5
[ 3302.054370] usb 1-1.4: failed to send control message: -19
[ 3302.057393] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 3302.057621] ch341 1-1.4:1.0: device disconnected
[ 3304.129333] usb 1-1.2: new full-speed USB device number 6 using xhci_hcd
[ 3304.225389] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 3304.225423] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 3304.225437] usb 1-1.2: Product: USB Serial
[ 3304.231482] ch341 1-1.2:1.0: ch341-uart converter detected
[ 3304.235702] usb 1-1.2: ch341-uart converter now attached to ttyUSB0
[ 3305.101582] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 3305.102074] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32