mukaschultze/ha-must-inverter #39: Add native support to other version Must PV (1/3/5/10kw)

Benj-HaRRison, 2024-09-21:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

This HA Integration successfully working for most of Must PV1800 models for watching to sensors. BUT this integration have have some limits for user input for another MUST devices (because this HA integration developed only to Must PV1800 3kW).

### Describe the solution you'd like

If you change MAXIMUM values x5 for all user inputs - your integration successfully will working with other MUST PV1800 devices to config too (not only for sensor monitoring)

### Describe alternatives you've considered

My device: Must 5kw, PV1800

### Additional context

Examples (this examples for MY device):

- ID "must_energy_inverter_battery_low_voltage" have maximum value "24", but other invertors have high of "24" - in my case i want set "42";
If you change maximum value to "85" - your integration will working with any PV1800.

- ID "must_energy_battery_stop_charging_voltage" have maximum value "29", but other invertors have high of "29" - in my case i want set "55,8";
If you change maximum value to "110" - your integration will working with any PV1800.

- ID "must_energy_battery_stop_discharging_voltage" have maximum value "29", but other invertors have high of "29" - in my case i want set "45,1";
If you change maximum value to "90" - your integration will working with any PV1800.

- ID "must_energy_float_voltage" have maximum value "29,2", but other invertors have high of "29,2" - in my case i want set "56,1";
If you change maximum value to "110" - your integration will working with any PV1800.

- ID "must_energy_battery_low_voltage" have maximum value "24", but other invertors have high of "24" - in my case i want set "34"
If you change maximum value to "70" - your integration will working with any PV1800.

- ID "must_energy_grid_max_charger_current_set" have maximum value "30", but other invertors have high of "30" - in my case i want set "60"
If you change maximum value to "120" - your integration will working with any PV1800.

p4block, 2024-10-31:
Yep, it's working great with my 5kW 48V inverter but all the sliders stop me from configuring it. It's in a quite remote location and well, it would be great if it let me set them :D 

I think the simplest way would be to let users configure the desired voltage when the integration is added?

mukaschultze, 2024-10-31:
> Yep, it's working great with my 5kW 48V inverter but all the sliders stop me from configuring it. It's in a quite remote location and well, it would be great if it let me set them :D
> 
> I think the simplest way would be to let users configure the desired voltage when the integration is added?

What's the model of your inverter? If you could paste the manual here it would be helpful.

The issue with fixing the voltages is that I don't know if they follow some sort of arithmetic progression or not. I don't know if the voltages for the 48V are exactly double of the 24V.

It doesn't need to be user-configured, the inverter itself reports whether it is 12V, 24V or 48V.

p4block, 2024-10-31:
@mukaschultze thanks for the quick response!

It's this one, MP1900
[image]

p4block, 2024-11-01:
@mukaschultze https://drive.google.com/file/d/1MYyRXOiXarxxhc5QiE7EMKxLuIbvSiIB/view?usp=drivesdk

I scanned the manual, hope it helps

mukaschultze, 2024-11-01:
> @mukaschultze [drive.google.com/file/d/1MYyRXOiXarxxhc5QiE7EMKxLuIbvSiIB/view?usp=drivesdk](https://drive.google.com/file/d/1MYyRXOiXarxxhc5QiE7EMKxLuIbvSiIB/view?usp=drivesdk)
> 
> I scanned the manual, hope it helps

Thank you very much @p4block, this helps a lot

Goolov, 2024-11-28:
Hello!
I use integration to connect Must PV18-5548 PREM.
The battery charging current of the specified inverter is from 0A-100A
But in the HA interface I can set 20A or 30A.
Is it possible to remove these restrictions in the HA interface?

sergeysaley, 2024-12-01:
Hi there!
I researched my Must PV19-6248 EXP and found several registers that might be interesting for your project 

To add:

battery state of charge (undocumented, comes from battery bms)
    address: 113
    unit_of_measurement: "%"
    register_type: holding

battery state of health (undocumented, comes from battery bms)
    address: 114
    unit_of_measurement: "%"
    register_type: holding

PV1 charger current (for older models - pv charger current)
    address: 15207
    register_type: holding
    unit_of_measurement: "A"
    accuracy_decimals: 1
    filters:
      multiply: 0.1

PV2 charger current (undocumented, but works)
    address: 16207
    register_type: holding
    unit_of_measurement: "A"
    accuracy_decimals: 1
    filters:
      multiply: 0.1

PV1 charger power (for older models - pv charger power)
    address: 15208
    register_type: holding
    unit_of_measurement: "W"
    device_class: power
    accuracy_decimals: 1

PV2 charger power (undocumented, but works)
    address: 16208
    register_type: holding
    unit_of_measurement: "W"
    device_class: power
    accuracy_decimals: 1

15208 and 16208 may be used as sources for Riemann sum sensors, useful in Energy Dashboard as Solar production sources.

You have the rest and almost everything works, so Must PV19-6248 EXP can be included in your compatibility list.
Only registers with errors and warnings do not work (n/a)

entvia, 2025-01-14:
@sergeysaley Thank you Serhii, I was looking for this info for several days, scanning registers :D 
I have PV19 EXP, but 4KW model. 

By the way, I have forked this repo and added option select PV1900, then all entities will be named by the model name and additonal registers will be applied. Will add your findings now. Maybe @mukaschultze would be interested to merge it later into this repo and we could potentially extend it with other devices and registers.

Thank you and thanks to @mukaschultze for this integration.

mukaschultze, 2025-01-14:
> @sergeysaley Thank you Serhii, I was looking for this info for several days, scanning registers :D I have PV19 EXP, but 4KW model.
>
> By the way, I have forked this repo and added option select [...]

Feel free to open a PR with your changes!

mukaschultze, 2025-01-14:
@entvia I reviewed your code and liked a lot your approach. When you have time please create the PR and I'll merge it into this repo.

One question that I have for you, does your `InverterSerialNumber` have a proper value? Mine and the ones from people that have sent me diagnostics all have 0xffffffff. Seems like the manufacturer never bothered to write the serial to the microcontroller memory. I saw that you added this field to the redact list in the diagnostics.

entvia, 2025-01-16:
Hi @mukaschultze Thank you for taking a look at my version, I've just created a pull request. 
I have the same issue with serial number, looks like Must doesn't set them :) Below is what I see: 

[image]

mukaschultze, 2025-02-03:
@entvia @sergeysaley (or anyone else that might have a pv1900) can you paste your diagnostics file here?

I'm trying to make the plugin auto-detect the PV1900 models instead of relying on the user configuration because the PV1800 can crash when reading unknown registers and I don't want the user to be able to select the wrong model.

entvia, 2025-02-05:
@mukaschultze Tried to download diagnostics, but getting the following error, so if you find some time, please take a look, otherwise I will try to check later.

```
Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
First occurred: 10:58:23 PM (5 occurrences)
Last logged: 11:00:10 PM
Error handling request

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/diagnostics/__init__.py", line 280, in get
    data = await info.config_entry_diagnostics(hass, config_entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/diagnostics.py", line 39, in async_get_config_entry_diagnostics
    devs_data = _async_devices_as_dict(hass, entry)
  File "/config/custom_components/must_inverter/diagnostics.py", line 55, in _async_devices_as_dict
    device_info = device._device_info()
                  ^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute '_device_info'
```

closed 2025-02-07 by mukaschultze