mukaschultze/ha-must-inverter #40: unable to connect to Inverter using TCP (WiFi-RS485)

LionBS76, 2024-09-28:
### System Health details

## System Information

version | core-2024.9.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.46-haos
arch | x86_64
timezone | Europe/Kyiv
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1422
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.09.1
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 30.8 GB
disk_used | 4.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Get HACS (1.3.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 18 вересня 2024 р. о 20:07
-- | --
current_recorder_run | 28 вересня 2024 р. о 16:21
estimated_db_size | 20.15 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Beforehand thanks for Your work!!! :)

I have invertor Must PV18-3224 that I want to monitor with HA.
HA and Invertor are on some distance from each other so I can't connect it directly by USB Serial cable.
To connect to Invertor remotely I use handmade WiFi-to-Serial adaptor (ESP-01 + esp-link software on it + rs485-to-UART adaptor).
I can use SolarPowerMonitor (origin software from Must) and [HW Virtual Serial Port software](https://www.hw-group.com/software/hw-vsp3-virtual-serial-port), on PC where SolarMonitor installed, to connect to my ESP and monitor Invertor.

So, my handmade adaptor shouldn't be the issue because origin soft SolarPowerMonitor is working fine.

I added Must Plugin to HA.
When I try to add Invertor device by TCP (use IP of ESP) that doesn't work.

My handmade WiFi-to-Serial like the one you used before (on DIN rail) in [your post](https://github.com/mukaschultze/ha-must-inverter/issues/8#issuecomment-2006859233) 

On the ESP (esp-link software) I can see some Debug.
When using Virtual Serial Port and origin Solar Monitor I see in log that soft make one connection (suppose 19200 baud) then close it. After that make one new connection and communicate through it all time.

Your HA Must plugin makes 5 connections and didn't close them. And Esp-link soft can't make more simultaneous connections.

 

### Reproduction steps

1. Connect WiFi-to-Serial to Must Invertor
2. Add Invrtor device by TCP option into HA Must plugin
3. Doesn't add any devices 
4. Using the same TCP info for connecting origin software Solar Power Monitor thiough Virtual Serial Port works fine
...

### Debug logs

```Text
From HA log:
2024-09-28 17:21:34.991 ERROR (MainThread) [custom_components.must_inverter] error reading modbus data at address 15201: Exception Response(166, 38, None)
2024-09-28 17:21:42.996 ERROR (MainThread) [custom_components.must_inverter] error reading modbus data at address 20000
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 239, in read_modbus_data
    response = await self._client.read_holding_registers(address=start, count=count, slave=0x04)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2024-09-28 17:21:46.010 ERROR (MainThread) [custom_components.must_inverter] error reading modbus data at address 20101
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 239, in read_modbus_data
    response = await self._client.read_holding_registers(address=start, count=count, slave=0x04)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 167, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
2024-09-28 17:21:46.013 ERROR (MainThread) [custom_components.must_inverter] error reading modbus data at address 25201
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 239, in read_modbus_data
    response = await self._client.read_holding_registers(address=start, count=count, slave=0x04)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 167, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
```

mukaschultze, 2024-09-28:
@LionBS76 what happens if you change the integration configuration so it doesn't reconnect too fast? Maybe the ESP can't handle the fast reconnect after the first error. Use something like in the screenshot below:

[image]

LionBS76, 2024-09-30:
> @LionBS76 what happens if you change the integration configuration so it doesn't reconnect too fast? Maybe the ESP can't handle the fast reconnect after the first error. Use something like in the [...]

I have tried such configuration parameters with no success.

LionBS76, 2024-10-02:
How can I get more debig for You to analyze ?