mukaschultze/ha-must-inverter #44: Extra sensors to easily integrate with the Energy Dashboard

p4block, 2024-11-03:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The energy dashboard for HA requests various sensors, some of which (I think) are not directly provided by the Must inverter. Namely, the battery kwh in and kwh out sensors.

### Describe the solution you'd like

The integration should do more things than read registers and present them to user, it should add these useful sensors that are not in the MPPT itself.

We have a sensor that has the instantaneous power in and out from the battery, with some magic it's possible to convert it to kwh in and out. 

### Describe alternatives you've considered

I made the sensors myself with chatpgt, but they are error prone. This is actually a thing that I encounter often and I always resort to some hacks, a well built solution coming from here would be best. 

### Additional context

In general, I found that the sensors provided by the inverter are not very good for human-readable consumption. Naming is very confusing and they hardly map to HA's (or my) needs. 

The battery one for the energy dashboard is the most obvious one, but I'm sure that thinking this through reveals many more.

mukaschultze, 2024-11-03:
I'm not entirely sure about adding "fake" sensors here — it could make the codebase overly complex. I know the sensors' naming is not the best, but I've tried to keep it as close to the official application as possible.

HomeAssistant already has a quite good integral sensor that can be easily set up via UI or YAML, I'd lean towards writing some sort of guide in the README file to instruct users on how the sensors work and how to create better kWh sensor than to add it directly in this integration.

That said, I understand the frustration with the kWh sensors provided by the device, I had to create my own DC power meter and coulomb counter using esphome because the inverter ones are quite useless and have low resolution and precision.

For reference, I've used these custom ones for some time:
```yaml
sensor:
  - platform: integration
    name: Battery Out
    unique_id: 'battery_out_kwh'
    source: sensor.battery_out
    unit_prefix: k
    unit_time: h
    round: 2
    method: left

  - platform: integration
    name: Battery In Solar
    unique_id: 'battery_in_solar_kwh'
    source: sensor.battery_in_solar
    unit_prefix: k
    unit_time: h
    round: 2
    method: left

  - platform: integration
    name: Solar Production
    unique_id: 'solar_production_kwh'
    source: sensor.pv1800_charger_power
    unit_prefix: k
    unit_time: h
    round: 2
    method: left
```

Where: 
- `sensor.battery_out` = `{{ max(0, states('sensor.pv1800_batt_power') | float) }}`
- `sensor.battery_in_solar` = `{{ -min(0, states('sensor.pv1800_batt_power') | float) }}`

p4block, 2024-11-03:
I agree that more documentation would be a good step, I've stared at these sensors for hours trying to make sense of what they truly are.  Thanks for your template sensors!

That said users will want to use this integration to fill their energy dashboard, having an almost mandatory manual "post-install" that could very easily be included seems counter productive. 

I also couldn't help but notice that you have solar production as charger_power, on my unit I use accumulated power. Maybe because my unit is entirely offgrid, but half the sensors always measure 0. 

[image]

In any case, if there's more information on where these sensors are and what they are measuring I will make sure to read it.

mukaschultze, 2024-11-03:
> I also couldn't help but notice that you have solar production as charger_power, on my unit I use accumulated power. Maybe because my unit is entirely offgrid, but half the sensors always measure 0.

@p4block Most of mine measures 0 as well.

> In any case, if there's more information on where these sensors are and what they are measuring I will make sure to read it.

From my experience:

- **buy power**: unknown, always zero
- **charger power**: DC power that went into the batteries (solar power only)
- **discharger power**: DC power out of the batteries
- **grid charger power**: unknown, always zero
- **load power**: total AC out (including bypass)
- **power**: total solar power
- **pv sell power**: unknown, always zero
- **self use power**: total AC out (not including bypass)

p4block, 2024-11-03:
@mukaschultze those are the assumptions I had initially, I will write down my thoughts:

**charger power**  goes to both battery and load. Thankfully we have standalone battery and inverter sensors for that. 

I arranged the three together in my dashboard to more or less make sense of what's happening.

[image]

In this picture taken  just before sunset it's getting 10W from panels (lol) +65W from battery are feeding the inverter which is outputting 24W.

[image]
After panels are off we see the bigger picture. The numbers don't match up, but that's easily explained, the device itself draws some tens of watts and the inverter is extremely inefficient at low loads. 
I'm measuring with an AC clamp just after its output and well, it's like below 50% efficiency when using 30ish W.

**discharger power** in theory should be the same as 
```
  - platform: integration
    name: Battery Out
    unique_id: 'battery_out_kwh'
    source: sensor.battery_out
    unit_prefix: k
    unit_time: h
    round: 2
    method: left
```
But I recall having problems with it, I will compare in a few days.  After the values repopulate. 

**load power** seems good but seemingly useless for the Energy Dashboard. It's also 0 for me, but it wasn't before I reset it. Weird.

**power** is good and I am feeding it directly as solar input in the dashboard

**self use power** is actually increasing since I reset it, unlike load_power. Maybe it has something to do with the offgrid status or some settings in the mppt. I have it set to SOL right now, but didn't in the past.

mukaschultze, 2024-11-03:
> After panels are off we see the bigger picture. The numbers don't match up, but that's easily explained, the device itself draws some tens of watts and the inverter is extremely inefficient at low [...]

Again, from my experience: 

The AC power sensor used in the inverter is super low precision for small loads, mine would sometimes report negative values when drawing less than 50W, above this threshold it correctly reported the same as a CT clamp. The solar power (charger power) and battery power were precise even for small loads.

The battery voltage sensor is absolute trash, with a slope error of ±1V. The solar panels' voltage was correct with only ±0.1V offset.

There are ways of "calibrating" the sensors, but I tried that and it doesn't make them any better.

My measured efficiency from battery power to AC power was about ~80%.

p4block, 2024-11-03:
I am drawing <10W, my unit seems to not fuck up that hard it goes into negative. It's not exactly the same, but efficiencies below 50% are not unheard of in computer PSUs at <10% load and I'm here asking a 5kW inverter for 0.01kW  

The clamp is currently off as I had to unplug half to house to conserve power, when I get it back up I can provide more accurate information.