mukaschultze/ha-must-inverter #47: WARNING (MainThread) [custom_components.must_inverter] wrong number of registers read at address *, expected *, got *

Benj-HaRRison, 2024-11-17:
### System Health details

## System Information

version | core-2024.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/Kyiv
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1460
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 129.2 GB
disk_used | 32.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | vmware
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Get HACS (1.3.1), File editor (5.8.0), Terminal & SSH (9.15.0), VLC (0.3.0), Advanced SSH & Web Terminal (19.0.0), FTP (5.1.1), Log Viewer (0.17.1), Spotify Connect (0.13.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
[... 142 lines cut ...]
2024-11-17 23:29:44.476 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 20000 to 20016 (17)
2024-11-17 23:29:44.476 INFO (MainThread) [custom_components.must_inverter] modbus client is not connected, trying to reconnect
2024-11-17 23:29:44.476 DEBUG (MainThread) [custom_components.must_inverter] connecting to /dev/ttyUSB0:0
2024-11-17 23:29:44.480 INFO (MainThread) [custom_components.must_inverter] successfully connected to /dev/ttyUSB0:0
2024-11-17 23:29:46.595 WARNING (MainThread) [custom_components.must_inverter] wrong number of registers read at address 20000, expected 17, got 21
2024-11-17 23:29:46.595 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 20101 to 20214 (114)
2024-11-17 23:29:48.914 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 25201 to 25279 (79)
2024-11-17 23:29:51.162 DEBUG (MainThread) [custom_components.must_inverter] finished reading modbus data, {10101: 1, 10102: 500, 10103: 561, 10104: 561, 10105: 440, 10106: 440, 10107: 600, 10108: 800, 10109: 100, 10110: 1, 10111: 280, 10112: 0, 10113: 48, 10114: 800, 10115: 56638, 10116: 0, 10117: 0, 10118: 0, 10119: 558, 10120: 150, 10121: 60, 10122: 120, 10123: 0, 10124: 0, 20101: 1, 20102: 2300, 20103: 5000, 20104: 0, 20105: 1, 20106: 0, 20107: 1, 20108: 0, 20109: 4, 20110: 0, 20111: 0, 20112: 0, 20113: 217, 20114: 217, 20115: 300, 20116: 300, 20117: 515, 20118: 525, 20119: 540, 20120: 1, 20121: 500, 20122: 561, 20123: 561, 20124: 48, 20125: 400, 20126: 100, 20127: 420, 20128: 600, 20129: 1, 20130: 200, 20131: 440, 20132: 1400, 20133: 0, 20134: 0, 20135: 0, 20136: 0, 20137: 0, 20138: 0, 20139: 0, 20140: 0, 20141: 0, 20142: 75, 20143: 3, 20144: 1, 20145: 0, 20146: 0, 20147: 200, 20148: 950, 20149: 0, 20150: 0, 20151: 0, 20152: 0, 20153: 0, 20154: 0, 20155: 0, 20156: 0, 20157: 0, 20158: 0, 20159: 0, 20160: 0, 20161: 65535, 20162: 0, 20163: 65535, 20164: 65535, 20165: 65535, 20166: 65535, 20167: 65535, 20168: 65535, 20169: 65535, 20170: 65535, 20171: 65535, 20172: 65535, 20173: 65535, 20174: 65535, 20175: 65535, 20176: 65535, 20177: 65535, 20178: 65535, 20179: 65535, 20180: 65535, 20181: 65535, 20182: 65535, 20183: 65535, 20184: 65535, 20185: 0, 20186: 0, 20187: 0, 20188: 0, 20189: 0, 20190: 0, 20191: 0, 20192: 0, 20193: 0, 20194: 0, 20195: 0, 20196: 0, 20197: 0, 20198: 0, 20199: 0, 20200: 0, 20201: 0, 20202: 0, 20203: 0, 20204: 0, 20205: 0, 20206: 0, 20207: 0, 20208: 0, 20209: 0, 20210: 0, 20211: 0, 20212: 0, 20213: 0, 20214: 0, 25201: 3, 25202: 230, 25203: 5000, 25204: 0, 25205: 529, 25206: 2343, 25207: 2335, 25208: 4246, 25209: 0, 25210: 0, 25211: 107, 25212: 106, 25213: 0, 25214: 63036, 25215: 2500, 25216: 49, 25217: 46, 25218: 2498, 25219: 2518, 25220: 0, 25221: 45, 25222: 120, 25223: 284, 25224: 0, 25225: 0, 25226: 4999, 25227: 0, 25228: 0, 25229: 0, 25230: 0, 25231: 0, 25232: 0, 25233: 31, 25234: 53, 25235: 31, 25236: 0, 25237: 1, 25238: 1, 25239: 1, 25240: 0, 25241: 1, 25242: 0, 25243: 0, 25244: 0, 25245: 0, 25246: 326, 25247: 0, 25248: 2429, 25249: 0, 25250: 3624, 25251: 0, 25252: 0, 25253: 0, 25254: 5744, 25255: 0, 25256: 2221, 25257: 0, 25258: 0, 25259: 0, 25260: 326, 25261: 0, 25262: 0, 25263: 0, 25264: 0, 25265: 0, 25266: 0, 25267: 0, 25268: 0, 25269: 65535, 25270: 65535, 25271: 10101, 25272: 21707, 25273: 0, 25274: 0, 25275: 48, 25276: 0, 25277: 5000, 25278: 10414, 25279: 750}
```

### Diagnostics dump

This error occurred in custom integration.
Logger: custom_components.must_inverter
Source: custom_components/must_inverter/__init__.py:243
integration: Must Inverter (documentation, problems)
First message: November 17, 2024 at 00:53:21 (25 messages)
Last message: November 17, 2024 at 11:29:46 pm

wrong number of registers read at address 15201, expected 21, got 24
wrong number of registers read at address 20000, expected 17, got 21
wrong number of registers read at address 20101,
expected 114, got 17
wrong number of registers read at address 20101, expected 114, got 21
wrong number of registers read at address 10101, expected 24, got 79

mukaschultze, 2024-11-20:
I couldn't find any code paths that would lead to this, I added that warning as a sanity check and it should never happen. I guess it's some bug on pymodbus,o n newer versions of it they added a check on their side that throws an exception when it happens

Benj-HaRRison, 2024-11-22:
> I couldn't find any code paths that would lead to this, I added that warning as a sanity check and it should never happen. I guess it's some bug on pymodbus,o n newer versions of it they added a check on their side that throws an exception when it happens

In other words, can I ignore the warnings?

Benj-HaRRison, 2024-11-22:
New logs:

2024-11-22 16:14:51.974 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data
2024-11-22 16:14:51.974 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 10101 to 10124 (24)
2024-11-22 16:14:52.181 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 15201 to 15221 (21)
2024-11-22 16:14:54.312 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 20000 to 20016 (17)
2024-11-22 16:14:56.436 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 20101 to 20214 (114)
2024-11-22 16:14:58.754 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 25201 to 25279 (79)
2024-11-22 16:15:01.001 DEBUG (MainThread) [custom_components.must_inverter] finished reading modbus data, {10101: 1, 10102: 500, 10103: 561, 10104: 561, 10105: 440, 10106: 440, 10107: 600, 10108: 800, 10109: 100, 10110: 1, 10111: 280, 10112: 0, 10113: 48, 10114: 800, 10115: 56638, 10116: 0, 10117: 0, 10118: 0, 10119: 558, 10120: 150, 10121: 60, 10122: 120, 10123: 0, 10124: 0, 15201: 2, 15202: 1, 15203: 1, 15204: 0, 15205: 557, 15206: 528, 15207: 0, 15208: 0, 15209: 29, 15210: 0, 15211: 1, 15212: 1, 15213: 0, 15214: 0, 15215: 48, 15216: 800, 15217: 0, 15218: 2879, 15219: 36, 15220: 13, 15221: 6, 20000: 20566, 20001: 1800, 20002: 65535, 20003: 65535, 20004: 10101, 20005: 21707, 20006: 10414, 20007: 65535, 20008: 65535, 20009: 65535, 20010: 65535, 20011: 65535, 20012: 65535, 20013: 65535, 20014: 65535, 20015: 65535, 20016: 65535, 20101: 1, 20102: 2300, 20103: 5000, 20104: 0, 20105: 1, 20106: 0, 20107: 1, 20108: 0, 20109: 4, 20110: 0, 20111: 0, 20112: 0, 20113: 217, 20114: 217, 20115: 300, 20116: 300, 20117: 515, 20118: 525, 20119: 540, 20120: 1, 20121: 500, 20122: 561, 20123: 561, 20124: 48, 20125: 200, 20126: 100, 20127: 420, 20128: 600, 20129: 1, 20130: 200, 20131: 440, 20132: 1400, 20133: 0, 20134: 0, 20135: 0, 20136: 0, 20137: 0, 20138: 0, 20139: 0, 20140: 0, 20141: 0, 20142: 75, 20143: 3, 20144: 1, 20145: 0, 20146: 0, 20147: 200, 20148: 950, 20149: 0, 20150: 0, 20151: 0, 20152: 0, 20153: 0, 20154: 0, 20155: 0, 20156: 0, 20157: 0, 20158: 0, 20159: 0, 20160: 0, 20161: 65535, 20162: 0, 20163: 65535, 20164: 65535, 20165: 65535, 20166: 65535, 20167: 65535, 20168: 65535, 20169: 65535, 20170: 65535, 20171: 65535, 20172: 65535, 20173: 65535, 20174: 65535, 20175: 65535, 20176: 65535, 20177: 65535, 20178: 65535, 20179: 65535, 20180: 65535, 20181: 65535, 20182: 65535, 20183: 65535, 20184: 65535, 20185: 0, 20186: 0, 20187: 0, 20188: 0, 20189: 0, 20190: 0, 20191: 0, 20192: 0, 20193: 0, 20194: 0, 20195: 0, 20196: 0, 20197: 0, 20198: 0, 20199: 0, 20200: 0, 20201: 0, 20202: 0, 20203: 0, 20204: 0, 20205: 0, 20206: 0, 20207: 0, 20208: 0, 20209: 0, 20210: 0, 20211: 0, 20212: 0, 20213: 0, 20214: 0, 25201: 3, 25202: 230, 25203: 5000, 25204: 0, 25205: 527, 25206: 2308, 25207: 2300, 25208: 4230, 25209: 0, 25210: 0, 25211: 72, 25212: 72, 25213: 0, 25214: 63924, 25215: 1630, 25216: 32, 25217: 0, 25218: 1645, 25219: 1647, 25220: 0, 25221: 0, 25222: 294, 25223: 200, 25224: 0, 25225: 0, 25226: 4999, 25227: 0, 25228: 0, 25229: 0, 25230: 0, 25231: 0, 25232: 0, 25233: 31, 25234: 54, 25235: 31, 25236: 0, 25237: 1, 25238: 1, 25239: 1, 25240: 0, 25241: 1, 25242: 0, 25243: 0, 25244: 0, 25245: 0, 25246: 373, 25247: 0, 25248: 2583, 25249: 0, 25250: 4267, 25251: 0, 25252: 0, 25253: 0, 25254: 6491, 25255: 0, 25256: 2353, 25257: 0, 25258: 0, 25259: 0, 25260: 373, 25261: 0, 25262: 0, 25263: 0, 25264: 0, 25265: 0, 25266: 0, 25267: 0, 25268: 0, 25269: 65535, 25270: 65535, 25271: 10101, 25272: 21707, 25273: 0, 25274: 0, 25275: 48, 25276: 0, 25277: 5000, 25278: 10414, 25279: 639}
2024-11-22 16:15:11.977 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data
2024-11-22 16:15:11.977 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 10101 to 10124 (24)

**2024-11-22 16:15:17.984 ERROR (MainThread) [custom_components.must_inverter] error reading modbus data at address 10101
Traceback (most recent call last):
  File "/config/custom_components/must_inverter/__init__.py", line 228, in read_modbus_data
    response = await self._client.read_holding_registers(address=start, count=count, slave=0x04)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 2 retries**

2024-11-22 16:15:17.985 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 15201 to 15221 (21)

**2024-11-22 16:15:17.985 INFO (MainThread) [custom_components.must_inverter] modbus client is not connected, trying to reconnect**
**2024-11-22 16:15:17.985 DEBUG (MainThread) [custom_components.must_inverter] connecting to /dev/ttyUSB0:0
2024-11-22 16:15:17.989 INFO (MainThread) [custom_components.must_inverter] successfully connected to /dev/ttyUSB0:0
2024-11-22 16:15:22.107 WARNING (MainThread) [custom_components.must_inverter] wrong number of registers read at address 15201, expected 21, got 24**

2024-11-22 16:15:22.107 DEBUG (MainThread) [custom_components.must_inverter] reading modbus data from 20000 to 20016 (17)
[... cut ...] 34: 54, 25235: 31, 25236: 0, 25237: 1, 25238: 1, 25239: 1, 25240: 0, 25241: 1, 25242: 0, 25243: 0, 25244: 0, 25245: 0, 25246: 373, 25247: 0, 25248: 2583, 25249: 0, 25250: 4267, 25251: 0, 25252: 0, 25253: 0, 25254: 6491, 25255: 0, 25256: 2353, 25257: 0, 25258: 0, 25259: 0, 25260: 373, 25261: 0, 25262: 0, 25263: 0, 25264: 0, 25265: 0, 25266: 0, 25267: 0, 25268: 0, 25269: 65535, 25270: 65535, 25271: 10101, 25272: 21707, 25273: 0, 25274: 0, 25275: 48, 25276: 0, 25277: 5000, 25278: 10414, 25279: 639}

mukaschultze, 2024-11-22:
> > I couldn't find any code paths that would lead to this, I added that warning as a sanity check and it should never happen. I guess it's some bug on pymodbus,o n newer versions of it they added a check on their side that throws an exception when it happens
> 
> In other words, can I ignore the warnings?

Yes, they should be harmless. I saw your other issue on the repo, it might be somewhat related to that. I would recommend increasing the scan interval configuration to something more moderate.

Benj-HaRRison, 2024-11-26:
> Yes, they should be harmless. I saw your other issue on the repo, it might be somewhat related to that. I would recommend increasing the scan interval configuration to something more moderate.

But scan interval in default value...

mukaschultze, 2025-02-11:
Closed by https://github.com/mukaschultze/ha-must-inverter/issues/47, possibly not related to this integration, but rather a race condition with the serial port when multiple devices are connected to the same inverter.