mukaschultze/ha-must-inverter #48: ValueError: filedescriptor out of range in select()

Benj-HaRRison, 2024-11-22:
### System Health details

## System Information

version | core-2024.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.54-haos
arch | x86_64
timezone | Europe/Kyiv
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1460
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 129.2 GB
disk_used | 15.1 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | vmware
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Get HACS (1.3.1), File editor (5.8.0), Terminal & SSH (9.15.0), VLC (0.3.0), Advanced SSH & Web Terminal (19.0.0), FTP (5.1.1), Log Viewer (0.17.1), Spotify Connect (0.13.0), Bambu Lab P1 - Spaghetti Detection Server (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 ноября 2024 г. в 15:29
-- | --
current_recorder_run | 21 ноября 2024 г. в 21:38
estimated_db_size | 198.54 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Every few days (from 12 hours to two days) the problem “ValueError: filedescriptor out of range in select()” appears steadily, with all sensors falling into the “Unavailable” status or sensor values ​​“freezing” at the last received data.

The problem cannot be reproduced by force; it appears randomly.
I didn’t notice any connections: the problem can appear at 9 pm, 5 am or 12 noon. In this case, no manipulations are performed: no one reconfigures anything in Home Assistant or touches the inverter.

The following messages appear in the HA log:
Регистратор: homeassistant
Источник: /usr/src/homeassistant/homeassistant/runner.py:147
Первое сообщение: 11:59:56 (4848551 сообщений)
Последнее сообщение: 15:45:02

Error doing job: Exception in callback SerialTransport._read_ready() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport_serial.py", line 128, in _read_ready
    if data := self.sync_serial.read(1024):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()

To restore work ha-must-inverter I need to restart Home Assistant every time and only then all the data is read again.
What helps is restarting Home Assistant:
1. Removing the cable from the USB socket of the inverter and reconnecting does not solve the problem
2. Removing the cable from the server’s USB socket and reconnecting does not solve the problem
3. Restarting the integration does not solve the problem
4. Replacing the USB cable does not solve the problem
5. Rebooting the inverter does not solve the problem

USB or Serial device path: /dev/ttyUSB0
Baud rate: 19200
Parity (‘E’ven, ‘O’dd or ‘N’one): N
Stop bits (1, 1.5, 2): 1
Byte size (7, 8): 8

Windows Server is used as the server,
on which the virtual machine VMware Workstation Pro 17.5.0 build-22583795 for Home Assistant is installed. Every time you turn on the virtual machine, the specified USB device is automatically forwarded to VmWare.

Interesting fact: on the Windows Server, a program from the manufacturer "Solar Monitor" is installed and when the problem is present,
If you remove forwarding of the USB device to the virtual machine, Solar Monitor in Windows Server works and reads data. If you forward the device back to VmWare, nothing will work until you restart Home Assistant.

### Reproduction steps

The problem cannot be reproduced by force; it appears randomly.
I didn’t notice any connections: the problem can appear at 9 pm, 5 am or 12 noon. In this case, no manipulations are performed: no one reconfigures anything in Home Assistant or touches the inverter.

### Debug logs

```Text
Регистратор: homeassistant
Источник: /usr/src/homeassistant/homeassistant/runner.py:147
Первое сообщение: 11:59:56 (4848551 сообщений)
Последнее сообщение: 15:45:02

Error doing job: Exception in callback SerialTransport._read_ready() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport_serial.py", line 128, in _read_ready
    if data := self.sync_serial.read(1024):
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/serial/serialposix.py", line 565, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()
```

### Diagnostics dump

[28.txt](https://github.com/user-attachments/files/17871440/28.txt)

p4block, 2024-11-23:
I can add something to this, I am running out of file descriptors on my system within the <24hrs roughly ever since I added this integration, I am not 100% sure the culprit is this integration.

Check out the output of `lsof -w | wc` and check out its output too `lsof -w > openfiles.txt`. The debug log of HA may also show problems with "too many open files"

mukaschultze, 2024-11-23:
@p4block are you also using HASS in a docker container?

mukaschultze, 2024-11-23:
I've updated pymodbus to 3.6.9, this is the same version that HASS core is using and it should be more stable for long lasting connections (There were multiple reports of issues in the core integration very similar to the ones I received here).

I recently moved and haven't had time to setup my inverter yet, so please let me know if this improves or worsens the situation. Thank you very much!

p4block, 2024-11-25:
@mukaschultze problem seems fixed after yesterday's update, fds don't creep up to the bajillions anymore

mukaschultze, 2024-12-03:
Closing this issue based on the feedback received here. Thank you all!

closed 2024-12-03 by mukaschultze