mukaschultze/ha-must-inverter #65: Reset & Lost Parameters after update ha-must-inverter

Benj-HaRRison, 2025-02-12:
### System Health details

## System Information

version | core-2024.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Kyiv
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1531
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 129.2 GB
disk_used | 9.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | vmware
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Get HACS (1.3.1), File editor (5.8.0), VLC (0.3.0), Advanced SSH & Web Terminal (20.0.0), Log Viewer (0.17.1), Spotify Connect (0.13.0), Bambu Lab P1 - Spaghetti Detection Server (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 1
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 27 января 2025 г. в 12:07
-- | --
current_recorder_run | 12 февраля 2025 г. в 05:49
estimated_db_size | 282.83 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After updating the HA-Must-Inverter to the latest version 2.0.0, the integration proposes to indicate the serial number of the device (to support several inverters). After indicating the serial number, a new device is added to the MUST list. The existing device on which stored history, indicators, etc. It becomes inactive. 

If you reconfigure integration and specify another device name or number there, a new device is added. All the old ones, although activated, are “inaccessible” with statuses. Only the last added device will monitor. For this reason, I was forced to roll back to the older version, which works more stable.

I am not against updating integration and adding a new device, but the whole story is attached to the old.
I can use the newly -incorporate device and everything works, only the story of this device as well as all automation, settings and bindings remain with the old one.

There is some opportunity to update integration,
Do not add a new device, but change the serial number of the old and use the old device?

### Reproduction steps

1. Update HA-must-Inverter to ver 2.0.0
2. Reload HA
3. Add serial number to your device

### Debug logs

```Text
None
```

mukaschultze, 2025-02-12:
I understand the frustration and confusion, but the whole point of the serial number repair is that HomeAssistant requires some sort of unique number for each device and for entities and the device doesn't provide one.

The only info the inverter provides that can be used for this identification is the serial number and the manufacturer didn't bother to write a proper value to it on almost all models.

The moment the serial number changes HomeAssistant itself creates a new device, that's how the device registry works. The alternative solution for this would be to have no device and leave the entities "floating" without a proper topology.

As for the broken automations and entity names, removing the config entry and re-adding it should fix the conflicting names.

mukaschultze, 2025-02-12:
Fixed by 9b5922b2f6852fdc6c51c4e4945ec41b8c887d4a

closed 2025-02-12 by mukaschultze

Benj-HaRRison, 2025-02-25:
> Fixed by [9b5922b](https://github.com/mukaschultze/ha-must-inverter/commit/9b5922b2f6852fdc6c51c4e4945ec41b8c887d4a)

mukaschultze, after testing last update my issue don't fixed :(

What i can do? I'm again used my backup, before update...
After update my old device unreachable, in device list I see a new device without history from old device...

Everyone who update this integration lost history of all sensors....