mukaschultze/ha-must-inverter #71: Switches does not work on latest Home Assistant

elik745i, 2025-04-17:
### System Health details

Core
2025.4.2
Supervisor
2025.04.0
Operating System
Frontend
20250411.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/mukaschultze/ha-must-inverter/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have successfully installed this integration, however I cant control it using switches, I get this error:
Failed to perform the action switch/turn_off. ModbusClientMixin.write_register() takes 3 positional arguments but 4 were given

### Reproduction steps

1.
2.
3.
...

### Debug logs

```Text
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 16:07:48 (5 occurrences)
Last logged: 16:16:25

[140731946226816] Unexpected exception
[140731775531568] Unexpected exception
[140731754319152] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2837, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1007, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1079, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/config/custom_components/must_inverter/switch.py", line 163, in async_turn_off
    return await self._async_set_value(False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/must_inverter/switch.py", line 156, in _async_set_value
    await self._inverter.write_modbus_data(ADDRESS, int(new_value))
  File "/config/custom_components/must_inverter/__init__.py", line 272, in write_modbus_data
    response = await self._client.write_register(address, value, 0x04)
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.write_register() takes 3 positional arguments but 4 were given
```

elik745i, 2025-04-19:
OK QUICK UPDATE:
After days of troubleshooting this is what needs to be changed to get it fully functional:
in init.py

for writing registers
change this:
response = await self._client.write_register(address, value,0x04)

to this:
response = await self._client.write_register(address=address, value=value, slave=0x04)

mukaschultze, 2025-05-07:
Closed by a5a2162d57ba90d3de3cb7d81b0ecbd0982227ca

closed 2025-05-07 by mukaschultze